IBO-GmbH/hsb_indicator_old #17: include guard

AleMatGra, 2024-11-27:
Using both #pragma once and #ifndef is redundant.

We should use one of those options.

I would suggest using #pragma once, it's less to write, compile times might go down. To be honest I think the general reluctance to use it is more of an emotional response to the idea that it _might_ cause issue rather than a logical one.

AndreasRichie, 2024-12-02:
I'm fine with either, it is possible to customize the extension "C++ Class Creator" to match whatever is chosen, although I don't think using only _#pragma once_ would reduce compile time in a noticeable way.
If @schaufelbagger also doesn't care I would suggest we go with the way @AleMatGra proposed and use only _#pragma once_

AndreasReichenauerIBO, 2025-01-20:
will be added to the coding guidelines and therefore to the wiki, so this issue will be closed

closed 2025-01-20 by AndreasReichenauerIBO