IBO-GmbH/hsb_indicator_old #26: check out LVGL

AlexanderGrathIBO, 2024-12-11:
Check out LVGL and if it is a viable alternative for SquareLine_Studio.

AndreasReichenauerIBO, 2024-12-16:
Did I understand it wrong or aren't those 2 different things? LVGL is the coding library and SquareLine_Studio is just a Software to do the design more easily and the design is then converted into code using LVGL

AlexanderGrathIBO, 2024-12-18:
> Did I understand it wrong or aren't those 2 different things? LVGL is the coding library and SquareLine_Studio is just a Software to do the design more easily and the design is then converted into code using LVGL

Added more info and context. LVGL is wat SquareLine_Studi uses in the back-end.
There are multiple issues with it, one of them is that they are no longer working together: https://forum.lvgl.io/t/lvgl-ends-its-collaboration-with-squareline-studio/14638
It does not support the new 9.x Version of LVGL.
In our code base, we use a port that uses code that no longer exists in the newest version, we are stuck with 8.3.X, any display revision requiring a code update would be a disaster.

AlexanderGrathIBO, 2025-02-19:
Issue can be closed, and [26](https://github.com/IBO-GmbH/hsb_indicator/tree/26-check-out-lvgl) can be removed. EEZ-Studio works well enough for UI creation for now. Manual changes can be made if needed, there is no point in using LVGL directly.

closed 2025-02-19 by AndreasReichenauerIBO